bigtreetech/CB2 #92: CB2 release V3.0.2 Wi-fi is not working.

Literav, 2025-04-30:
CB2 release V3.0.2 firmware cb2_debian12_klipper_ernel6.1_20250429.img .
Wi-fi is not working.

CB2 release V 1.0.3 firmware cb2_debian11_klipper_ernel4.19_20240422.img.xz
WI-FI is working fine.

hellEVIL-13, 2025-05-09:
> CB2 release V3.0.2 firmware cb2_debian12_klipper_ernel6.1_20250429.img . Wi-fi is not working.
> 
> CB2 release V 1.0.3 firmware cb2_debian11_klipper_ernel4.19_20240422.img.xz WI-FI is working fine.

WI-FI is working fine on BTT PI V2

bigtreetech, 2025-05-13:
> CB2 release V3.0.2 firmware cb2_debian12_klipper_ernel6.1_20250429.img . Wi-fi is not working.
> 
> CB2 release V 1.0.3 firmware cb2_debian11_klipper_ernel4.19_20240422.img.xz WI-FI is working fine.

@Literav Make sure that your settings in the cfg file are correct. The WiFi driver is working fine on all of our bench test units as well as our beta testers.

bigtreetech, 2025-05-13:
> I entered the network name and password correctly. Wi-fi works on version 1,0.3, but not on version 3.0.2 and the rest. Which driver to watch where? Can I send diagnostic files if possible? I changed Routers, took the old D-link DIR-300. I tried distributing WI-FI from a Samsung S24ultra phone. I bought a K-Touch from you and it works great with my WI-FI network.

Does your wifi password use a '$' character?

Literav, 2025-05-15:
No, just the numbers,

Literav, 2025-05-21:
Hello! 
In CB2 release V 1.0.3, the WI-FI settings in the config file are fragmented.
CB2 release V3.0.2 the wi-fi settings in the config file are commented out. 
I didn't pay attention. I spelled out everything correctly, but I forgot to remove the "#" sign.
The topic can be closed.Thank you all!!!

closed 2025-05-21 by Literav

bigtreetech, 2025-05-21:
Good to hear.

iamLellep, 2025-06-03:
hi, i have the same problem, wifi does not connect with the same credentials of the old iso.
I was careful to uncomment the right sections but nothing. I also tried to use '!' for the special character but nothing changed.
I have not been selling any for a month. Also not even with a network cable can you see the network

[image]

bigtreetech, 2025-06-03:
We generated a hotspot using a local iPhone with the exact settings you have mentioned and were able to connect instantly. Please use notepad++ to perform edits to this file and not the standard notepad app. We don't trust it even if it says that it is only using unix line endings.

[image]

iamLellep, 2025-06-03:
> We generated a hotspot using a local iPhone with the exact settings you have mentioned and were able to connect instantly. Please use notepad++ to perform edits to this file and not the standard notepad app. We don't trust it even if it says that it is only using unix line endings.
> 
> [image]

thank you for the answer. I tried now with notepad++ but the result is the same. I also lowered the protection from wpa2/wpa3 to wpa/wpa2, but nothing changed.
It seems very strange to me that with the iso 1.0.5 everything works fine but already from 2.0 it no longer connects.
So I exclude a hardware problem.
Please help me, otherwise I will have to stay with the old version or even worse replace with a cm4

bigtreetech, 2025-06-03:
Also, do you get an IP address when you plug directly into Ethernet?

iamLellep, 2025-06-03:
> Are you sure that your module is even booting up using the latest image? If you get nothing on the screen and no IP via ethernet then it would seem that it is probably not even booting. Are you trying to run of SD or eMMC?

I'm starting to have doubts that the cb2 does not boot, even if the green led flashes.
I'm using the same sdcard for both iso 1.0.5 and 3.0.2. Have you changed anything about booting from sdcard?

bigtreetech, 2025-06-03:
I think it is very much something with your particular setup. This ISO is already running on thousands of modules without these issues. I would start by checking your SD card. Could be faulty.

Mister-Erik, 2026-02-17:
Hello, 
I have had the same issues with writing to the emmc en not working WiFi with armbian 3.0.2.