bigtreetech/CB2 #95: Communication Timeout over USB

cjackson234, 2025-05-07:
This issue seems to be related to https://github.com/bigtreetech/CB2/issues/88

When I home it works, but every few tries QGL/Tilt Adjust fails.  There are modifications that can be made to the klipper mcu.py that actually get it up and running most of the time. But I get failures occasionally still. Sometime it requires a reboot to fix.

Hardware:

- Nitehawk SB
- Manta M8P
- CB2
- Beacon Probe

Please note: I have the exact same hardware in another printer with a CM4, so this does seem to be related to the CB2 in some way.

bigtreetech, 2025-05-07:
Could you post the logs of a failure?

cjackson234, 2025-05-07:
Error during homing contact: Communication timeout during homing
Stats 12012.2: gcodein=0  mcu: mcu_awake=0.002 mcu_task_avg=0.000002 mcu_task_stddev=0.000001 bytes_write=10238888 bytes_read=3094616 bytes_retransmit=9 bytes_invalid=0 send_seq=220404 receive_seq=220404 retransmit_seq=2 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=399994037 CB2: mcu_awake=0.001 mcu_task_avg=0.000012 mcu_task_stddev=0.000015 bytes_write=45882 bytes_read=143467 bytes_retransmit=0 bytes_invalid=0 send_seq=7619 receive_seq=7619 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=49999853 adj=50000569 nhk: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000004 bytes_write=3216267 bytes_read=1428391 bytes_retransmit=9 bytes_invalid=0 send_seq=65205 receive_seq=65205 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=11999839 adj=12000038  MCU: temp=38.6 SoC: temp=38.9 heater_bed: target=0 temp=39.1 pwm=0.000 beacon: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=127259 bytes_read=22731597 bytes_retransmit=0 bytes_invalid=0 send_seq=14384 receive_seq=14384 retransmit_seq=0 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=31999623 adj=32000096 coil_temp=38.2 refs=2 mcu_temp=40.24 supply_voltage=3.033 nh_temp: temp=55.0 sysload=2.04 cputime=2436.262 memavail=1501212 print_time=7403.686 buffer_time=0.091 print_stall=0 extruder: target=0 temp=42.4 pwm=0.000
Stats 12013.2: gcodein=0  mcu: mcu_awake=0.002 mcu_task_avg=0.000002 mcu_task_stddev=0.000001 bytes_write=10239034 bytes_read=3094949 bytes_retransmit=9 bytes_invalid=0 send_seq=220415 receive_seq=220415 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=399994056 CB2: mcu_awake=0.001 mcu_task_avg=0.000012 mcu_task_stddev=0.000015 bytes_write=45888 bytes_read=143483 bytes_retransmit=0 bytes_invalid=0 send_seq=7620 receive_seq=7620 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=49999848 adj=50000956 nhk: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000004 bytes_write=3216273 bytes_read=1428527 bytes_retransmit=9 bytes_invalid=0 send_seq=65206 receive_seq=65206 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=11999841 adj=12000036  MCU: temp=38.4 SoC: temp=39.4 heater_bed: target=0 temp=39.0 pwm=0.000 beacon: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=127265 bytes_read=22734673 bytes_retransmit=0 bytes_invalid=0 send_seq=14385 receive_seq=14385 retransmit_seq=0 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=31999625 adj=32000087 coil_temp=38.2 refs=2 mcu_temp=40.26 supply_voltage=3.033 nh_temp: temp=54.9 sysload=2.03 cputime=2436.524 memavail=1500512 print_time=7403.686 buffer_time=0.000 print_stall=0 extruder: target=0 temp=42.4 pwm=0.000
Stats 12014.2: gcodein=0  mcu: mcu_awake=0.002 mcu_task_avg=0.000002 mcu_task_stddev=0.000001 bytes_write=10239180 bytes_read=3095297 bytes_retransmit=9 bytes_invalid=0 send_seq=220426 receive_seq=220426 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=399994087 CB2: mcu_awake=0.001 mcu_task_avg=0.000012 mcu_task_stddev=0.000016 bytes_write=45894 bytes_read=143512 bytes_retransmit=0 bytes_invalid=0 send_seq=7621 receive_seq=7621 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=49999852 adj=50000772 nhk: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000004 bytes_write=3216279 bytes_read=1428650 bytes_retransmit=9 bytes_invalid=0 send_seq=65207 receive_seq=65207 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=11999841 adj=12000052  MCU: temp=38.4 SoC: temp=38.9 heater_bed: target=0 temp=39.0 pwm=0.000 beacon: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=127271 bytes_read=22737749 bytes_retransmit=0 bytes_invalid=0 send_seq=14386 receive_seq=14386 retransmit_seq=0 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=31999628 adj=32000099 coil_temp=38.2 refs=2 mcu_temp=40.27 supply_voltage=3.033 nh_temp: temp=54.9 sysload=2.03 cputime=2436.797 memavail=1500272 print_time=7403.686 buffer_time=0.000 print_stall=0 extruder: target=0 temp=42.5 pwm=0.000
Stats 12015.2: gcodein=0  mcu: mcu_awake=0.002 mcu_task_avg=0.000002 mcu_task_stddev=0.000001 bytes_write=10239326 bytes_read=3095659 bytes_retransmit=9 bytes_invalid=0 send_seq=220437 receive_seq=220437 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=399994081 CB2: mcu_awake=0.001 mcu_task_avg=0.000012 mcu_task_stddev=0.000016 bytes_write=45900 bytes_read=143528 bytes_retransmit=0 bytes_invalid=0 send_seq=7622 receive_seq=7622 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=49999851 adj=50000725 nhk: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000004 bytes_write=3216285 bytes_read=1428803 bytes_retransmit=9 bytes_invalid=0 send_seq=65208 receive_seq=65208 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=11999841 adj=12000039  MCU: temp=38.7 SoC: temp=39.4 heater_bed: target=0 temp=39.0 pwm=0.000 beacon: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=127277 bytes_read=22740825 bytes_retransmit=0 bytes_invalid=0 send_seq=14387 receive_seq=14387 retrans [... cut ...] eacon: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=127295 bytes_read=22750053 bytes_retransmit=0 bytes_invalid=0 send_seq=14390 receive_seq=14390 retransmit_seq=0 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=31999624 adj=32000014 coil_temp=38.3 refs=2 mcu_temp=40.35 supply_voltage=3.034 nh_temp: temp=54.9 sysload=2.11 cputime=2437.805 memavail=1501768 print_time=7410.559 buffer_time=0.000 print_stall=0 extruder: target=0 temp=42.3 pwm=0.000

hellEVIL-13, 2025-05-10:
I have the same problem, only this helps sudo systemctl stop KlipperScreen.service

jfml2020, 2025-05-25:
Caution: I've had problems with disconnections and loss of MCU communication, etc.
You need to pay attention to the power contacts on the BTT Pi2 or the boards on the printer in the soldering area.
I've had print stops twice. Once, it was due to poor contact with the probes, and the second time, it was due to soldering on an SKR 3 mini board. Likewise, the power soldering contacts.

They randomly generate power supply voltages that cause disconnections.

Hooba97, 2025-05-29:
i am facing a similar issue with the new kernel version 

i am using Manta m8p v2 + CB2 + EBB SB2209 USB + Eddy Duo USB  

I tried to investigate why there is a bytes_retransmit=9 across all MCUs at boot 
but it doesn't look like the cause of this 

most probably its related to Multiple Micro-controller Homing and Probing timing

[klippy (1).log](https://github.com/user-attachments/files/20511778/klippy.1.log)