bigtreetech/CB2 #100: When using RKDevTool download boot Fail

rhiz0matic, 2025-05-12:
I've tried to flash the EMMC uninstalling and reinstalling the rockusb driver, resetting unplugging and reloading the board, and i get the error:P
download boot fail
Note: please check ddr, please reset device and retry

[image]

jfml2020, 2025-05-13:
> He intentado flashear el EMMC desinstalando y reinstalando el controlador rockusb, reiniciando, desconectando y volviendo a cargar la placa, y me aparece el error: P error de arranque de descarga. Nota: verifique el ddr, reinicie el dispositivo y vuelva a intentarlo.
> 
> [image]

is it in otg mode...? is that sd card formatter?

bigtreetech, 2025-05-13:
You must ensure that the motherboard it is installed on has had the dip switches placed according to the programming requirements in the motherboard manual. There are two dip switches that need to be adjusted. You also need to ensure that the USB port on the motherboard is configured for direct access to the CB2 and not for UART to USB conversion. Some motherboards have a separate dip switch close to the USB to configure this. All info will be available in the manual for your motherboard.