bigtreetech/CB2 #101: Wi-Fi Not working V3.0.2 on cb2

RcPaz, 2025-05-17:
Using V3.0.2 on my cb2 board my wi-fi will not connect. 

When using v1.0.5 I can connect no issues at all. 

The cb2 board is connected to a Manta m5p board.

Literav, 2025-05-21:
Hi! 
In version CB2 V 1.0.3, the WI-FI settings in the configuration file are commented out
In CB2 V3.0.2, the wi-fi settings in the configuration file are commented out. 
I wasn't paying attention. I stated everything correctly, but I forgot to remove the "#" sign.

closed 2025-05-21 by bigtreetech

RcPaz, 2025-05-23:
I have done all that, there is no special characters in my password, I have tried numerous times, and I cannot get v3.0.2 to work.
as soon as I flash V1.0.3 it works immediately. 

I run a cb1 board on my manta m4p board that works no problem 
I have multiple raspberry Pi boards which have no issues

all are running klipper

Literav, 2025-05-23:
It should be as below.It should be as below. Example of a section from config

## The delimiter for a string is ''.
## If the string contains a ', please use spliced fragments to implement it.
## Other characters (such as double quotes["], spaces[ ], $, etc.) can be used directly without escaping.
## That is to say, ' needs to be replaced with '\''
## For example, if the password is: 12$34 56"78'9
## Need to be set as: WIFI_SSID='12$34 56"78'\''9'
## wifi name
WIFI_SSID='ZYIPTest'
## wifi password
WIFI_PASSWD='12345678'

Literav, 2025-05-23:
Remove the # before WIFI_SSID='ZYIPTest'
Remove the # before WIFI_PASSWD='12345678'

Compare config 1.0.3 and 3.0.2 in the wifi part.
Two files side by side and you'll understand what I mean.

RcPaz, 2025-05-23:
yeah i have, I have even tried copying over the working config from v1.0xxx (which does connect) and replacing the config file in V3.0.2 and still nothing, it cant be a hardware issue because it does work when i flash the early img file

RcPaz, 2025-06-05:
sorry for the late reply, i no longer have the cb2 and m5p board i gave it away to a friend, he also could not get it to work on V3.0.2

RcPaz, 2025-06-12:
got the cb2 and manta board back, after a few failed attempts to boot img V3.0.2 from an Sd card, I tried a new Sd card and still had no luck, when I tried flashing img 1.0.5 it worked straight away as previously mentioned

I ended up connecting to the cb2 board through RKDEV tool and erased the Emmc and was still not able to get V3.0.2 to bootup off a Sd card. I ended up flashing the latest img to the directly to the  Emmc using RKDEV and was able to get V3.0.2 to boot up and is now working. (currently installed in a manta m4p board which replaced my cb1)

 
I believe the whole issue I was having was a boot up problem