bigtreetech/CB2 #104: How to update properly?

igl-x2, 2025-06-03:
Situation is:
Pi2 is mounted inside printer case. Its a pita to get it out.
System is installed on eMMC.
Access to the Pi2 is either Wifi/ssh or USB (via USB-A 2.0).

Anyway to update to version 3.0.2 without opening the case and messing around with sd cards etc?

bigtreetech, 2025-06-04:
Can update over USB but need to flick dip switches on Pi accordingly. Manual and wiki have all the details.

valik1017, 2025-06-23:
Hello!

But if it's on BIGTREETECH MANTA M8P V2.0, how to update?

Thank you!

valik1017, 2025-06-23:
[image]

RKDevTool not woking, or should be this?

valik1017, 2025-06-23:
[image]

I think you are have some errors in wiki for CB2 or answer from AI is wrong? )

bigtreetech, 2025-06-23:
AI is wrong. Recovery button is on the CB2. Please check the CB2 manual.

AlexTheGreat1971, 2025-06-30:
Will everything be erased during updating (printer's cfgs, programms ets)?