bigtreetech/CB2 #111: Bottom USB 2.0 Port Fails to Enumerate Devices via Powered Hub

kanishk619, 2025-08-02:
### Summary

USB 3.2 industrial powered hub not working correctly on **bottom USB 2.0 port**. Devices connected to the hub are not detected, while the same setup works perfectly on the **top USB 2.0 port**.

### Details
- **Board**: Manta M8P V2.0 + CB2
- **OS**: Armbian 3.0.2 (Latest)
- **Hub**: [Powered USB 3.2 hub](https://www.waveshare.com/usb3.2-gen1-hub-4u.htm)
- **Top port**: Hub + all devices work
- **Bottom port**: Hub detected, devices not

### `lsusb -t` (bottom port)

```bash
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 4: Dev 7, If 0, Class=Communications, Driver=cdc_acm, 12M
        |__ Port 4: Dev 7, If 1, Class=CDC Data, Driver=cdc_acm, 12M
```

### `lsusb -t` (top port)

```bash
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 12, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 1: Dev 13, If 0, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 1: Dev 13, If 1, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 4: Dev 14, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
        |__ Port 4: Dev 7, If 0, Class=Communications, Driver=cdc_acm, 12M
        |__ Port 4: Dev 7, If 1, Class=CDC Data, Driver=cdc_acm, 12M
```

kanishk619, 2025-08-19:
@bigtreetech any inputs ?

AcrimoniousMirth, 2026-01-22:
Appear to be experiencing the same.