bigtreetech/CB2 #118: Brand new CB2 V3.0.2 halting on its own after a fresh install

rjandaya, 2025-09-05:
Just received a CB2 to replace my CB1. It is currently paired with a Pi4B Adapter.
Flashed the CB2 3.0.2 firmware without Obico (CB2_Debian12_Klipper_kernel6.1_20250821_without_obico.img.xz) via RKDevTool

All I've done is update the Klipper packages in the Mainsail UI. After a reboot, the CB2 runs fine, from the logs, after ~10min, the system halts on its own.  It seems that 16:36:34 is the timestamp where the issue begins.

```
biqu@BTT-CB2:~$ journalctl -f
Sep 04 16:27:36 BTT-CB2 KlipperScreen-start.sh[2784]: 20250904 16:27:36,404 [main_menu.py:add_device] - Adding device: temperature_sensor CB1
Sep 04 16:27:36 BTT-CB2 KlipperScreen-start.sh[2784]: 20250904 16:27:36,421 [main_menu.py:add_device] - Adding device: temperature_sensor MCU
Sep 04 16:27:36 BTT-CB2 KlipperScreen-start.sh[2784]: 20250904 16:27:36,431 [main_menu.py:add_device] - Adding device: temperature_sensor chamber
Sep 04 16:27:36 BTT-CB2 KlipperScreen-start.sh[2784]: 20250904 16:27:36,441 [main_menu.py:add_device] - Adding device: temperature_sensor dryer
Sep 04 16:27:36 BTT-CB2 KlipperScreen-start.sh[2784]: 20250904 16:27:36,486 [menu.py:arrangeMenuItems] - X > gcodes
Sep 04 16:27:36 BTT-CB2 KlipperScreen-start.sh[2784]: 20250904 16:27:36,512 [screen.py:attach_panel] - Current panel hierarchy: main_menu
Sep 04 16:27:36 BTT-CB2 KlipperScreen-start.sh[2784]: 20250904 16:27:36,550 [printer.py:init_temp_store] - Temp store: ['heater_bed', 'temperature_sensor CB1', 'temperature_sensor MCU', 'temperature_sensor chamber', 'temperature_sensor dryer', 'extruder']
Sep 04 16:27:36 BTT-CB2 KlipperScreen-start.sh[2784]: 20250904 16:27:36,567 [screen.py:init_tempstore] - Temperature store size: 1200
Sep 04 16:27:36 BTT-CB2 PackageKit[1816]: daemon quit
Sep 04 16:27:36 BTT-CB2 systemd[1]: packagekit.service: Deactivated successfully.
Sep 04 16:28:34 BTT-CB2 KlipperScreen-start.sh[2784]: 20250904 16:28:34,695 [base_panel.py:battery_percentage] - Battery information not available.
Sep 04 16:30:01 BTT-CB2 CRON[2926]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Sep 04 16:30:01 BTT-CB2 CRON[2927]: (root) CMD (/usr/lib/armbian/armbian-truncate-logs)
Sep 04 16:30:01 BTT-CB2 CRON[2926]: pam_unix(cron:session): session closed for user root
Sep 04 16:33:21 BTT-CB2 sshd[3101]: Accepted password for biqu from 10.10.0.1 port 61022 ssh2
Sep 04 16:33:21 BTT-CB2 sshd[3101]: pam_unix(sshd:session): session opened for user biqu(uid=1000) by (uid=0)
Sep 04 16:33:21 BTT-CB2 systemd-logind[779]: New session 7 of user biqu.
Sep 04 16:33:21 BTT-CB2 systemd[1]: Started session-7.scope - Session 7 of User biqu.
Sep 04 16:33:23 BTT-CB2 sshd[3101]: pam_env(sshd:session): deprecated reading of user environment enabled
Sep 04 16:36:34 BTT-CB2 systemd[1]: Starting fstrim.service - Discard unused blocks on filesystems from /etc/fstab...
Sep 04 16:36:35 BTT-CB2 fstrim[3529]: /boot: 86.5 MiB (90730496 bytes) trimmed on /dev/mmcblk1p1
Sep 04 16:36:35 BTT-CB2 fstrim[3529]: /: 22.7 GiB (24421474304 bytes) trimmed on /dev/mmcblk1p2
Sep 04 16:36:35 BTT-CB2 systemd[1]: fstrim.service: Deactivated successfully.
Sep 04 16:36:35 BTT-CB2 systemd[1]: Finished fstrim.service - Discard unused blocks on filesystems from /etc/fstab.
Sep 04 16:37:19 BTT-CB2 systemd[1]: Starting systemd-tmpfiles-clean.service - Cleanup of Temporary Directories...
Sep 04 16:37:19 BTT-CB2 systemd[1]: systemd-tmpfiles-clean.service: Deactivated successfully.
Sep 04 16:37:19 BTT-CB2 systemd[1]: Finished systemd-tmpfiles-clean.service - Cleanup of Temporary Directories.
Sep 04 16:37:19 BTT-CB2 systemd[1]: run-credentials-systemd\x2dtmpfiles\x2dclean.service.mount: Deactivated successfully.
Sep 04 16:40:36 BTT-CB2 KlipperScreen-start.sh[2765]: /usr/bin/xinit: connection to X server lost
Sep 04 16:40:36 BTT-CB2 KlipperScreen-start.sh[2765]: 
Sep 04 16:40:36 BTT-CB2 python[1344]: [websockets.py:on_close()] - Websocket Closed: ID: 281473192183568 Close Code: None, Close Reason: None, Pong Time Elapsed: 21.44
Sep 04 16:40:37 BTT-CB2 KlipperScreen-start.sh[2765]: waiting for X server to shut down
Sep 04 16:40:37 BTT-CB2 KlipperScreen-start.sh[2773]: (II) Server terminated successfully (0). Closing log file.
Sep 04 16:40:37 BTT-CB2 (sd-pam)[2769]: pam_unix(root:session): session closed for user biqu
Sep 04 16:40:37 BTT-CB2 systemd[1]: KlipperScreen.service: Deactivated successfully.
Sep 04 16:40:37 BTT-CB2 systemd[1]: session-c2.scope: Deactivated successfully.
Sep 04 16:40:37 BTT-CB2 systemd[1]: session-c2.scope: Consumed 2min 10.312s CPU time.
Sep 04 16:40:38 BTT-CB2 systemd-logind[779]: Session c2 logged out. Waiting for processes to exit.
Sep 04 16:40:38 BTT-CB2 systemd-logind[779]: Removed session c2.
Sep 04 16:40:40 BTT-CB2 systemd[1]: KlipperScreen.service: Scheduled restart job, restart counter is at 2.
Sep 04 16:40:40 BTT-CB2 systemd[1]: Stopped KlipperScreen.service - KlipperScreen.
Sep 04 16:40:40 BTT-CB2 systemd[1]: Started KlipperScreen.service - KlipperScreen.
Sep 04 16:40:40 BTT-CB2 (start.sh)[3732]: pam_unix(root:session): session opened for user biqu(uid=1000) by biqu(uid=0)
Sep 04 16:40:40 BTT-CB2 systemd-logind[779]: New session c3 of user biqu.
Sep 04 16:40:40 BTT-CB2 systemd[1]: Started session-c3.scope - Session c3 of User biqu.
Sep 04 16:40:40 BTT-CB2 KlipperScreen-start.sh[3732]: Running KlipperScreen on X in display :0 by default
Sep 04 16:40:40 BTT-CB2 KlipperScreen-start.sh[3738]: X.Org X Server 1.21.1.7
Sep 04 16:40:40 BTT-CB2 KlipperScreen-start.sh[3738]: X Protocol Version 11, Revision 0
Sep 04 16:40:40 BTT-CB2 KlipperScreen-start.sh[3738]: Current Operating System: Linux BTT-CB2 6.1.115-btt-rk35xx #1 SMP Mon Apr 28 18:57:36 CST 2025 aarch64
Sep 04 16:40:40 BTT-CB2 KlipperScreen-start.sh[3738]: Kernel command line: root=UUID=afa8801a-614f-4f56-87c7-82c002fdc465 rootwait rootfstype=ext4 splash=verbose console=ttyS2,1500000 console=tty1 consoleblank=0 loglevel=1 ubootpart=43282925-01 usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u cma=256M  cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory
Sep 04 16:40:40 BTT-CB2 KlipperScreen-start.sh[3738]: xorg-server 2:21.1.7-3+deb12u9 (https://www.debian.org/support)
Sep 04 16:40:40 BTT-CB2 KlipperScreen-start.sh[3738]: Current version of pixman: 0.42.2
Sep 04 16:40:40 BTT-CB2 KlipperScreen-start.sh[3738]:         Before reporting problems, check http://wiki.x.org
Sep 04 16:40:40 BTT-CB2 KlipperScreen-start.sh[3738]:         to make sure that you have the latest version.
Sep 04 16:40:40 BTT-CB2 KlipperScreen-start.sh[3738]: Markers: (--) probed, (**) from config file, (==) default setting,
Sep 04 16:40:40 BTT-CB2 KlipperScreen-start.sh[3738]:         (++) from command line, (!!) notice, (II) informational,
Sep 04 16:40:40 BTT-CB2 KlipperScreen-start.sh[3738]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Sep 04 16:40:40 BTT-CB2 KlipperScreen-start.sh[3738]: (==) Log file: "/var/log/Xorg.0.log", Time: Thu Sep  4 16:40:40 2025
Sep 04 16:40:40 BTT-CB2 KlipperScreen-start.sh[3738]: (==) Using config directory: "/etc/X11/xorg.conf.d"
Sep 04 16:40:40 BTT-CB2 KlipperScreen-start.sh[3738]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Sep 04 16:40:43 BTT-CB2 python[3742]: AT-SPI: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
Sep 04 16:40:44 BTT-CB2 KlipperScreen-start.sh[3742]: 20250904 16:40:44,141 [functions.py:log_start] - --------------------KlipperScreen Log Start--------------------
Sep 04 16:40:44 BTT-CB2 KlipperScreen-start.sh[3742]: 20250904 16:40:44,142 [functions.py:log_start] - KlipperScreen Version: v0.4.6-11-g6b6f63b
[... 60 lines cut ...]
Sep 04 16:40:46 BTT-CB2 KlipperScreen-start.sh[3742]: 20250904 16:40:46,145 [KlippyWebsocket.py:object_subscription] - Sending printer.objects.subscribe
Sep 04 16:40:46 BTT-CB2 KlipperScreen-start.sh[3742]: 20250904 16:40:46,149 [files.py:set_gcodes_path] - Gcodes path: /home/biqu/printer_data/gcodes
Sep 04 16:40:46 BTT-CB2 KlipperScreen-start.sh[3742]: 20250904 16:40:46,150 [screen.py:init_klipper] - Printer initialized
Sep 04 16:40:46 BTT-CB2 KlipperScreen-start.sh[3742]: 20250904 16:40:46,152 [printer.py:change_state] - Changing state from 'disconnected' to 'ready'
Sep 04 16:40:46 BTT-CB2 KlipperScreen-start.sh[3742]: 20250904 16:40:46,152 [printer.py:change_state] - Adding callback for state: ready
[... cut ...]
Sep 04 16:40:46 BTT-CB2 KlipperScreen-start.sh[3742]: 20250904 16:40:46,730 [screen.py:attach_panel] - Current panel hierarchy: main_menu
Sep 04 16:40:46 BTT-CB2 KlipperScreen-start.sh[3742]: 20250904 16:40:46,776 [printer.py:init_temp_store] - Temp store: ['heater_bed', 'temperature_sensor CB1', 'temperature_sensor MCU', 'temperature_sensor chamber', 'temperature_sensor dryer', 'extruder']
Sep 04 16:40:46 BTT-CB2 KlipperScreen-start.sh[3742]: 20250904 16:40:46,796 [screen.py:init_tempstore] - Temperature store size: 1200
Connection to 10.10.0.102 closed by remote host.
Connection to 10.10.0.102 closed.
```

geekussupreme, 2025-09-08:
@rjandaya which image did you use?

rjandaya, 2025-09-08:
> [@rjandaya](https://github.com/rjandaya) which image did you use?

I used B2_Debian12_Klipper_kernel6.1_20250821_without_obico.img
I found that the wiki is out of date. In the root of this repo, the manual includes this section if V3.0.1 or newer was already installed, stating to use SD card formatter to erase the eMMC.

[image]

After doing that, I booted the CB2 without an SD card to ensure the eMMC was erased, and I assumed so when the activity light on my Pi4B stayed solid/dim, instead of flashing. 
I removed the power cable, booted from the SD card, updated klipper components and let the CB2 idle for ~15min see if it would halt on SD card boot as well, which it didn't.
I flashed the eMMC with SD card instructions and let the CB2 idle again before using it like normal

rjandaya, 2025-09-10:
> Is it still working as it should? 

so far so good